ros-visualization/rqt_common_plugins #300: Failed to process package 'rqt_image_view'

chien800614, 2015-01-21:
Scanning dependencies of target rqt_image_view
[ 14%] [ 28%] Building CXX object CMakeFiles/rqt_image_view.dir/src/rqt_image_view/image_view.cpp.o
Building CXX object CMakeFiles/rqt_image_view.dir/include/rqt_image_view/moc_image_view.cxx.o
/Users/chienyuwu/src/rqt_common_plugins/rqt_image_view/src/rqt_image_view/image_view.cpp:285:54: error: 
      use of undeclared identifier 'CV_GRAY2RGB'
        cv::cvtColor(cv_ptr->image, conversion_mat_, CV_GRAY2RGB);
                                                     ^
/Users/src/rqt_common_plugins/rqt_image_view/src/rqt_image_view/image_view.cpp:303:54: error: 
      use of undeclared identifier 'CV_GRAY2RGB'
        cv::cvtColor(img_scaled_8u, conversion_mat_, CV_GRAY2RGB);
                                                     ^
2 errors generated.
make[2]: *** [CMakeFiles/rqt_image_view.dir/src/rqt_image_view/image_view.cpp.o] Error 1
make[1]: *** [CMakeFiles/rqt_image_view.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'rqt_image_view':

dirk-thomas, 2015-01-21:
I just built the package without a problem (ROS Indigo on Ubuntu Trusty). Please provide more information about your platform and the version of the package itself as well as its dependencies.

chien800614, 2015-01-21:
Indigo in OSX10.10
-- Using PYTHON_EXECUTABLE: /System/Library/Frameworks/Python.framework/Versions/2.7/bin/python
-- Using default Python package layout
-- Using empy: /Library/Python/2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /Users/build_isolated/rqt_image_view/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.6.9
CMake Warning (dev) at /Users/install_isolated/share/qt_gui_cpp/cmake/qt_gui_cppConfig.cmake:100 (elseif):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

dirk-thomas, 2016-12-20:
Closing due to no response.

closed 2016-12-20 by dirk-thomas