ros-visualization/rqt_common_plugins #301: rqt_plot: remove warning/debug message when backend not found

trainman419, 2015-01-26:
New users are confused by the log messages indicating that some of the rqt_plot backends are not found. These debug messages should be removed from the console log, and a message indicating that the backends aren't installed should be added to the backend selection dialog.

I've seen this 3 or 4 times in as many months on ROS answers; the most recent occurrence is: http://answers.ros.org/question/201721/rxplot-not-found/?answer=201784#post-id-201784

Enfenion, 2015-02-09:
I second this. Or the message should at least state that another backend was found and and is used.

ablasdel, 2015-02-09:
I agree that warnings to the command line do not seem appropriate here as long as it finds something to use. The GUI for changing the backend already has instructions on what to install if you want to change them.

Would anyone be against removing these messages entirely? 

At the very least changing them to "INFO" level instead of warning and adjusting the message to get the point across that everything is still AOK would be a great change.

dirk-thomas, 2015-02-09:
I don't think simply removing the messages is a good idea. Simply because then there is no indicator anywhere what happened.

Having  some indication in the backend selection dialog sounds like the best approach to me. Does anyone want to provide a PR with that enhancement?

ablasdel, 2015-02-16:
The backend selection GUI already disables selection of the backends it can't find and points users to the installer of the backends in question.

This seems like enough to me. 
What more do you think would be warranted?

dirk-thomas, 2015-02-16:
You are right, the settings dialog already provide a lot of feedback. I am just a bit concerned that it isn't very discoverable (which is a problem for all plugins).

What do you think - would the following be helpful?
Instead of the warning the plugin outputs an info message to the console stating which backend it has selected. Additional the message mentions that this can be changed in the plugins settings dialog.

ablasdel, 2015-02-16:
I think that makes sense but I would still downgrade it to an info level message not a warning since this will be displaying every time they launch unless they have all the backends installed.

trainman419, 2015-02-16:
As a GUI tool, I think all of the user's interaction should be through the GUI.

It sounds like the settings menu already covers the required behavior - it informs the user which back ends are available, and how to install them if they aren't found.

In that case, I think we should remove the command-line warning entirely.

I don't think the settings menu being discoverable is an issue here - once the user has discovered the option to change the back end, they're given all of the appropriate information for that context.

@dirk-thomas: if you think the settings menu is not discoverable enogh, that should be filed and handled as a separate bug.

dirk-thomas, 2015-02-16:
Sounds good. Go ahead and remove the console warnings then.

commit bfebd581 (2015-02-17): Remove warning when backend not found. Fixes #301