ros-visualization/rqt_common_plugins #303: rqt_top: failure to launch

ablasdel, 2015-01-27:
noticed this while testing a pull request and don't have time to look at why right now but I get this when I try to load rqt_top from script. loading it through the plugin menu works.

```
Traceback (most recent call last):
  File "./rqt_top", line 34, in <module>
    roslib.load_manifest(pkg)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslib/launcher.py", line 62, in load_manifest
    sys.path = _generate_python_path(package_name, _rospack) + sys.path
  File "/opt/ros/hydro/lib/python2.7/dist-packages/roslib/launcher.py", line 93, in _generate_python_path
    m = rospack.get_manifest(pkg)
  File "/usr/lib/pymodules/python2.7/rospkg/rospack.py", line 159, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/pymodules/python2.7/rospkg/rospack.py", line 198, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/pymodules/python2.7/rospkg/rospack.py", line 190, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: rqt_top.top_plugin.Top
ROS path [0]=/opt/ros/hydro/share/ros
ROS path [1]=/home/ablasdel/groovy_caktin_ws/src
ROS path [2]=/opt/ros/hydro/share
ROS path [3]=/opt/ros/hydro/stacks
```

dirk-thomas, 2015-01-28:
Looking at the `rqt_top` script (https://github.com/ros-visualization/rqt_common_plugins/blob/a59fb6350852fe1035737130ad0ac59e0e8d3bfd/rqt_top/scripts/rqt_top#L31-L34) the usage of `roslib` is definitely wrong. In a catkin package that should be done.

It looks like it was introduced here: https://github.com/ros-visualization/rqt_common_plugins/commit/df9b875697e73f1998e51a1e8ffd7f6441708d3e#diff-7631a871b50b8a3c302ebdc9e0259ad0

ablasdel, 2015-01-29:
Yep I'll have time to test this early next week.

commit 6922222e (2015-02-13): Fixes #303 failure to launch rqt_top when loading from script
closed 2015-02-13 by ablasdel