ros-visualization/rqt_common_plugins #305: Array in SubMessage not shown in raw-view

TheGreatArt, 2015-02-26:
This issue is that i use

OuterMessage:
Header header
ClusterMsg[] clusters

ClusterMsg:
uint8[] indices
string label

But the content inside the raw view looks like in the image.
If i play the bagfile and listen with "rostopic echo /labels" i can see that the array is filled correctly

[image]

greetings!

dirk-thomas, 2015-03-04:
The raw view in `rqt_bag` tries to determine the field type based on the Python type (https://github.com/ros-visualization/rqt_common_plugins/commit/6640ce2b4347f33afeba0520c1fed6d26e2cfdf6#diff-09908728fca83b95e445d7e4438e4773R193). This is simply not feasible since the logic for that is **very** complex. Instead it should use functions provided by `genmsg` or `genpy` to determine the type based on the actual message definition.

@ablasdel @trainman419 May be one of you guys will find some time to look into this. A good starting point might be to look how `rostopic echo` does it (see https://github.com/ros/ros_comm/blob/1df83085f88313bfe3cc6510690726939b157772/tools/rostopic/src/rostopic/__init__.py#L575).

ablasdel, 2015-03-04:
This seems like a very important bug to fix but at the moment I don't have any time to show rqt any love.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_bag#4

closed 2017-04-24 by dirk-thomas