ros-visualization/rqt_common_plugins #306: rqt_bag: when moving a viewer, it cannot be resized anymore

suryaambrose, 2015-03-20:
Open any viewer (raw for instance). The viewer opens on the right of rqt_bag. It can be resized, the viewer will adapt. 
Grab the viewer and move it to the bottom of the window (or anywhere else). Then try to resize it => it won't move. If you bring it back to its original position, it will still not resize.

ablasdel, 2015-03-20:
Wow that is a serious flaw. I'm surprised I didn't ever run into it. 

Does it only happen with the bottom Or is it any other snapped location?
Does this happen every time?
Are you running in stand alone mode? If not are there any other plugins running?

This is how to reproduce correct?
Open rqt_bag
Open raw viewer
Reposition raw viewer to snapped in below rqt_bag
Try to resize raw viewer and you will be unable to.

suryaambrose, 2015-03-20:
Any other snapped location and yes it happens all the time. 
I run rqt_bag in stand alone mode and your protocol is correct :).

suryaambrose, 2015-03-20:
The bug is not that old (september 2014). 

Here is a picture where you can see that the viewer did not resize when I moved it at the bottom. This behavior will remain even if I move the viewer back to the right.
[image]

This would be the expected behavior
[image]

ablasdel, 2015-03-20:
Ok if you can figure out how to fix it it would be awesome if you could submit a patch. If not I'll take a look when I can get some time (not sure when that will be at the moment. )

Does moving the raw viewer out to a floating window after it gets stuck fix it?

If not for now the work around is Don't do that! ^_^

suryaambrose, 2015-03-20:
Hum I don't know about your question, I did not try, but I just submitted a patch anyway :)

ablasdel, 2015-03-20:
Are you a wizard?

Ok I'll try to get some time to review this. The new work around is use the pull request! 

Thanks suryaambrose!

trainman419, 2015-03-29:
I've confirmed that I can reproduce this.

trainman419, 2015-04-02:
I think this is fixed now that #307 is merged.

closed 2015-04-02 by ablasdel