ros-visualization/rqt_common_plugins #308: rqt_bag crashes when doing anything at all

ntraft, 2015-03-24:
On Hydro, Ubuntu 12.04, and python 2.7.3-0ubuntu2.2, when trying to record or load a bag file, I get the following:

	➜  .ros  rqt_bag finite_wall_2015-03-23-19-28-25.bag 
	Loading finite_wall_2015-03-23-19-28-25.bag
	Exception in thread Thread-8:
	Traceback (most recent call last):
	  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
	    self.run()
	  File "/usr/lib/python2.7/threading.py", line 504, in run
	    self.__target(*self.__args, **self.__kwargs)
	  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/bag.py", line 61, in load_bags
	    self._widget.load_bag(bagfile)
	  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/bag_widget.py", line 267, in load_bag
	    self._timeline.add_bag(bag)
	  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 167, in add_bag
	    self._timeline_frame.reset_timeline()
	  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 916, in reset_timeline
	    self.emit_play_region()
	  File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 256, in emit_play_region
	    self.scene().selected_region_changed.emit(*self.play_region)
	TypeError: BagTimeline.selected_region_changed[Time, Time].emit(): argument 1 has unexpected type 'NoneType'

ablasdel, 2015-03-24:
To ensure that the bag file isn't corrupt can you verify that you can play the bag file back using [Rosbag](http://wiki.ros.org/rosbag)?

if it works fine that way can you provide a copy of the bag and a list of steps to reproduce the issue?

Thanks

ntraft, 2015-03-24:
Or we could try this instead—steps to reproduce:
* Open rqt_bag: `$ rqt_bag`
* Select "Record Bag" toolbar button
* Type a name, e.g. "foobar". Hit the "Save" button.
* Apparently nothing happens, and you get this stack trace in the shell:

    [INFO] [WallTime: 1427189113.664990] Recording to /home/ntraft/.ros/foobar_2015-03-24-02-25-13.bag.
    Traceback (most recent call last):
      File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/bag_widget.py", line 235, in _handle_record_clicked
        self._timeline.record_bag(record_filename)
      File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 643, in record_bag
        self.add_bag(self._recorder.bag)
      File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 167, in add_bag
        self._timeline_frame.reset_timeline()
      File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 916, in reset_timeline
        self.emit_play_region()
      File "/opt/ros/hydro/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 256, in emit_play_region
        self.scene().selected_region_changed.emit(*self.play_region)
    TypeError: BagTimeline.selected_region_changed[Time, Time].emit(): argument 1 has unexpected type 'NoneType'

ablasdel, 2015-03-24:
As long as we have steps it works for me! 
I'm not on 12.04 or hydro I will see if i can reproduce when I get some time. 

It seems you have double checked but if you don't mind:
If you are using the debians can you triple check you have latest of rqt, qt_gui_core & rqt_common_plugins packages? 
Does this happen every time or is it intermittent? 
Do you have access to another computer? Is it showing the same behavior?

ntraft, 2015-03-24:
I have all the latest versions according to `apt-cache policy` (that command works, right?)

It happens every time. I'll try it on a different machine when I get the chance.

ablasdel, 2015-04-07:
Sorry for the delay! I have tested this on indigo/Trusty and I can't reproduce it.

Have you had a chance to check if this is a problem on any other machines?

ntraft, 2015-04-07:
Yes, same result on another Hydro/Precise machine.

ablasdel, 2015-04-07:
Hmm ok I don't know when I will get any time on a Hydro/Precise machine.
Perhaps later this week.

Just to make sure nobody is blocking on this bug please go here if you need to bag NOW: http://wiki.ros.org/rosbag

ablasdel, 2015-04-09:
Ok I replicated this on a fresh 12.04 vm with desktop_full from debs. 
Next I checked out the source and tried it from there on the same machine. 
The source checkout worked fine. Can you try this for you and see if it works on your setup?

Not sure when this issue was fixed but it seems it was after the most recent precise/hydro build?
Perhaps a new release would fix this?

ablasdel, 2015-04-09:
Got a few minutes to look again at why we have a difference Hydro vs Indigo. The Hydro/precise debs have rqt_bag version 0.3.9 while the latest version is 0.3.10 (indigo has this version).

Since the hydro buildfarm isn't running anymore I don't think the debs will be getting this upgrade. It seems "use this from source" might be the suggestion until you can upgrade.

Note the [ros distribution timeline](http://wiki.ros.org/Distributions/Timeline) suggests stopping use of  Hydro/Precise this month in favor of Trusty/Indigo. Any chance you can put a distro upgrade on your schedule?

Hey @dirk-thomas the hydro build farm is down forever right now? There is no way to do a one off deb build to replace an old deb correct?

dirk-thomas, 2015-04-09:
*Hydro* is not yet EOL and the build farm is still running as usual. It will be officially EOLed when Jade is being released. Commonly the farm will be around a while longer after that (but don't rely on it).

Anyway it sounds like a Hydro release would solve the problem from this ticket. @ablasdel You can either do a Hydro bloom-release now to the latest version also out to Hydro or do a new patch release to all active ROS distros soon if enough changes have accumulated in the repo.

ablasdel, 2015-04-09:
@dirk-thomas that is great! It looks like the new 0.3.10 version was [accidentally only released to Indigo](https://github.com/ros-gbp/rqt_common_plugins-release/blob/master/tracks.yaml#L32).

I'll work on releasing a 0.3.11 version and get it out for hydro & indigo.

ntraft, 2015-04-09:
Thanks Aaron!

ablasdel, 2015-05-04:
@dirk-thomas was gracious enough to release version 3.11. You can check rqt_bag here: http://www.ros.org/debbuild/hydro.html?q=rqt_bag

Once this sync happens this will all be resolved. Thanks again.

closed 2015-05-04 by ablasdel