ros-visualization/rqt_common_plugins #312: rqt_bag: plot plugin only lets you plot 1 topic

ablasdel, 2015-04-21:
It would be nice if you could get multiple topics in the same plugin graph for correlation.

trainman419, 2015-04-21:
The best way to do this would be through the same context menu that allows plotting: instead of just offering to plot a topic, the menu should also allow the user to add the selected topic to an existing plot.

ablasdel, 2015-04-21:
I like that. All we would have to do is give each of the port windows a unique title and display that as an option in the drop down. This actually doesn't sound super hard at all. If I get some time next week I will see if I can implement this before I release rqt.

ablasdel, 2016-04-20:
Coming up on the self assign anniversary... It is pretty clear other priorities won out. If that looks like it will change I will reassign myself.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_bag#7

closed 2017-04-24 by dirk-thomas