ros-visualization/rqt_common_plugins #320: rqt_bag publishes wrong topic after scrolling

Daniel-Eckert, 2015-05-12:
If I right click and publish in rqt_bag, everything works fine as long as I scroll to the top of the topic list. As soon as I scroll down, a different topic than the one I actually click on is published. It's like the scrolling works visually, but if I click on a topic it always takes the one that would have been there if I never scrolled down.
Operating system is Ubuntu 14.04.

ablasdel, 2015-05-12:
Are you saying you pause the playback and scroll and then restart play and it changes the publisher? or just scrolling causes publishers to go crazy?

Please list the steps I need to take to reproduce.

stuglaser, 2015-05-12:
I've seen this too.

When you scroll the list of topics, the right click is associated with the incorrect topic.  Instead of publishing or showing the topic under the mouse, it publishes or shows whatever topic was in that location when the scrollbar is at the top.

To reproduce:
- Run rqt_bag on a bag where the number of topics is larger than the window
- Scroll down the topics list
- Right click on a topic and try to view it
- The wrong topic shows up

I expect that rqt_bag fails to take into account scrolling when determining which topic is under the mouse cursor.

ablasdel, 2015-05-12:
Looks like the "map_y_to_topic" function doesn't take scrolling into account.

It uses the "screen Y" and instead of translating it into "Scene Y"

The function we want here is this one: http://doc.qt.io/qt-5/qgraphicsview.html#mapToScene-5

bricerebsamen, 2015-11-17:
I gave it a try. Basically calling `mapToScene` (I also tried `mapFromScene`) in this function does not do anything: `y` has same value after and before this call: `y = self.mapToScene(0, y).y()`

Maybe related: the scrollbar disappears or does not resize from time to time, the window's width needs to resized to bring it back or make it adjust. there are some notes in `_resizeEvent`.

reinzor, 2016-04-04:
What is the status of this issue"?

guillaumeautran, 2016-04-13:
I'm using version 0.3.12 of rqt_bag and are still seeing the issue. As soon as you scroll the list of topics, right click still target a different topic. 
For example, if the window shows the first 10 topics out of 20, everything works fine, however, if you scroll down, right clicking on the topic selects the topic that would have been there if you did not scroll. 
Hope this helps.

ablasdel, 2016-04-13:
@reinzor I unfortunately haven't had a chance to give this a look. If you can figure it out please go for it!

reinzor, 2016-04-14:
Submitted a quick fix as PR:

https://github.com/ros-visualization/rqt_common_plugins/pull/362

I think there should be a more pretty solution but I don't have enough knowledge about the structure at the moment.

guillaumeautran, 2016-04-14:
I'm looking at your PR reinzor. Will let you know if it works for me as well. Thanks for the prompt action here.

guillaumeautran, 2016-04-14:
@reinzor Your version has an extra 'plot' entry in the context menu 'view' submenu. Don't know what it's for but it hangs rqt_bag if selected...
As for the origin problem reported here, your fix makes it work properly. So, thanks!

reinzor, 2016-04-14:
@guillaumeautran , what do you mean with the extra 'plot' entry? I only changed the input op the map_y_to_topic function:

topic = self._timeline_frame.map_y_to_topic(self.views()[0].mapToScene(event.pos()).y())

guillaumeautran, 2016-04-14:
[image]

guillaumeautran, 2016-04-14:
Definitely not related to your change but since I did not have that option in the menu before and it does seem to hang rqt_bag, I just wanted to make sure it is not a regression for a new feature...

ablasdel, 2016-04-14:
Plot has existed for awhile now. It should be acceptable to ignore it in the context of this issue. Please create another issue for plot related hanging.

This mapping looks like what should be happing with a map to scene call. I haven't had a chance to test this. Has it solved this for you guilla?

guillaumeautran, 2016-04-14:
Sounds good. Yes, your fix solves my usecase. Thanks.

ablasdel, 2016-04-14:
Fix is merged

closed 2016-04-14 by ablasdel