ros-visualization/rqt_common_plugins #321: rqt_graph prints KeyError 'bb'

ablasdel, 2015-05-13:
```
PluginHandlerDirect._restore_settings() plugin "rqt_graph/RosGraph#0" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 116, in _restore_settings
    self._plugin.restore_settings(plugin_settings_plugin, instance_settings_plugin)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 202, in restore_settings
    self._refresh_rosgraph()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 226, in _refresh_rosgraph
    self._update_graph_view(self._generate_dotcode())
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 259, in _update_graph_view
    self._redraw_graph_view()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 292, in _redraw_graph_view
    same_label_siblings=True)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_dotgraph/dot_to_qt.py", line 254, in dotcode_to_qt_items
    subgraph_nodeitem = self.getNodeItemForSubgraph(subgraph, highlight_level)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_dotgraph/dot_to_qt.py", line 83, in getNodeItemForSubgraph
    bb = subgraph.attr['bb'].strip('"').split(',')
KeyError: 'bb'
```

This seems to be the same thing experienced in this ros answers question: http://answers.ros.org/question/192782/rqt_graph-crashes/

Unfortunately re-installing rqt_graph doesn't seem to fix it for us after installing 0.3.11.

rqt-graph & rqt-common-plugins
```
$ dpkg -l | grep rqt-graph
ii  ros-indigo-rqt-graph                                  0.3.11-0trusty-20150506-0526-+0000                  amd64        rqt_graph provides a GUI plugin for visualizing the ROS computation graph. Its components are made generic so that other packages where you want to achieve graph representation can depend upon this pkg (use rqt_dep to find out the pkgs that depend. rqt_dep itself depends on rqt_graph too).
$ dpkg -l | grep rqt-common
ii  ros-indigo-rqt-common-plugins                         0.3.11-0trusty-20150506-1410-+0000                  amd64        rqt_common_plugins metapackage provides ROS backend graphical tools suite that can be used on/off of robot runtime. To run any rqt plugins, just type in a single command &quot;rqt&quot;, then select any plugins you want from the GUI that launches afterwards. rqt consists of three following metapackages: rqt - core modules of rqt (ROS GUI) framework. rqt plugin developers barely needs to pay attention to this metapackage. rqt_common_plugins (you're here!) rqt_robot_plugins - rqt plugins that are particularly used with robots during their runtime.
```

content of the rqt_gui.ini file of the computer in question

```
[General]
perspectives=@@rqt_graph__RosGraph, @@rqt_plot__Plot, @@rqt_console__Console, @@rqt_logger_level__LoggerLevel, Default
current-perspective=Default

[plugin_manager]
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\qt_gui\rqt_web=/opt/ros/indigo/share/rqt_web/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\qt_gui\rqt_gui_cpp=/opt/ros/indigo/share/rqt_gui_cpp/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\qt_gui\rqt_shell=/opt/ros/indigo/share/rqt_shell/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\qt_gui\rqt_py_console=/opt/ros/indigo/share/rqt_py_console/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\qt_gui\rqt_dep=/opt/ros/indigo/share/rqt_dep/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\qt_gui\qt_gui_cpp=/opt/ros/indigo/share/qt_gui_cpp/plugin.xml
7517170443106272166\discovery_data\qt_gui_cpp.RosPluginlibPluginProvider\qt_gui%20qt_gui_cpp%3A%3APlugin=/opt/ros/indigo/share/qt_gui_cpp/plugin.xml, /opt/ros/indigo/share/rqt_shell/plugin.xml, /opt/ros/indigo/share/rqt_dep/plugin.xml, /opt/ros/indigo/share/rqt_py_console/plugin.xml, /opt/ros/indigo/share/rqt_web/plugin.xml, /opt/ros/indigo/share/rqt_gui_cpp/plugin.xml
7517170443106272166\discovery_data\qt_gui_cpp.RosPluginlibPluginProvider\qt_gui%20qt_gui_cpp%3A%3APluginProvider=/opt/ros/indigo/share/qt_gui_cpp/plugin.xml, /opt/ros/indigo/share/rqt_shell/plugin.xml, /opt/ros/indigo/share/rqt_dep/plugin.xml, /opt/ros/indigo/share/rqt_py_console/plugin.xml, /opt/ros/indigo/share/rqt_web/plugin.xml, /opt/ros/indigo/share/rqt_gui_cpp/plugin.xml
7517170443106272166\discovery_data\qt_gui_cpp.RosPluginlibPluginProvider\rqt_gui%20rqt_gui_cpp%3A%3APlugin=/opt/ros/indigo/share/rqt_launch/plugin.xml, /opt/ros/indigo/share/rqt_service_caller/plugin.xml, /opt/ros/indigo/share/rqt_pose_view/plugin.xml, /opt/ros/indigo/share/rqt_plot/plugin.xml, /opt/ros/indigo/share/rqt_msg/plugin.xml, /opt/ros/indigo/share/rqt_srv/plugin.xml, /opt/ros/indigo/share/rqt_publisher/plugin.xml, /opt/ros/indigo/share/rqt_runtime_monitor/plugin.xml, /opt/ros/indigo/share/rqt_image_view/plugin.xml, /opt/ros/indigo/share/rqt_top/plugin.xml, /opt/ros/indigo/share/rqt_reconfigure/plugin.xml, /opt/ros/indigo/share/rqt_topic/plugin.xml, /opt/ros/indigo/share/rqt_robot_steering/plugin.xml, /opt/ros/indigo/share/rqt_logger_level/plugin.xml, /opt/ros/indigo/share/rqt_graph/plugin.xml, /opt/ros/indigo/share/rqt_rviz/plugin.xml, /opt/ros/indigo/share/rqt_nav_view/plugin.xml, /opt/ros/indigo/share/rqt_console/plugin.xml, /opt/ros/indigo/share/rqt_tf_tree/plugin.xml, /opt/ros/indigo/share/rqt_moveit/plugin.xml, /opt/ros/indigo/share/rqt_gui_py/plugin.xml, /opt/ros/indigo/share/rqt_robot_monitor/plugin.xml, /opt/ros/indigo/share/rqt_bag/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_image_view=/opt/ros/indigo/share/rqt_image_view/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_graph=/opt/ros/indigo/share/rqt_graph/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_console=/opt/ros/indigo/share/rqt_console/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_runtime_monitor=/opt/ros/indigo/share/rqt_runtime_monitor/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_srv=/opt/ros/indigo/share/rqt_srv/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_reconfigure=/opt/ros/indigo/share/rqt_reconfigure/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_bag=/opt/ros/indigo/share/rqt_bag/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_plot=/opt/ros/indigo/share/rqt_plot/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_publisher=/opt/ros/indigo/share/rqt_publisher/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_robot_steering=/opt/ros/indigo/share/rqt_robot_steering/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_nav_view=/opt/ros/indigo/share/rqt_nav_view/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_launch=/opt/ros/indigo/share/rqt_launch/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_topic=/opt/ros/indigo/share/rqt_topic/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_action=/opt/ros/indigo/share/rqt_action/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_top=/opt/ros/indigo/share/rqt_top/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_gui_py=/opt/ros/indigo/share/rqt_gui_py/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_rviz=/opt/ros/indigo/share/rqt_rviz/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_tf_tree=/opt/ros/indigo/share/rqt_tf_tree/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_moveit=/opt/ros/indigo/share/rqt_moveit/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_service_caller=/opt/ros/indigo/share/rqt_service_caller/plugin.xml
7517170443106272166\discovery_data\rqt_gui.RospkgPluginProvider\rqt_gui\rqt_msg=/opt/ros/indigo/share/rqt_msg/plugin.xml
[... 50 lines cut ...]
%40rqt_console__Console\pluginmanager\plugin__rqt_console__Console__0\plugin\highlight_filter_0\regex=false
[... cut ...]
Default\pluginmanager\plugin__rqt_graph__RosGraph__1\plugin\leaf_topics_check_box_state=true
Default\pluginmanager\plugin__rqt_graph__RosGraph__1\plugin\quiet_check_box_state=true
Default\pluginmanager\plugin__rqt_graph__RosGraph__1\plugin\auto_fit_graph_check_box_state=true
Default\pluginmanager\plugin__rqt_graph__RosGraph__1\plugin\highlight_connections_check_box_state=true
Default\pluginmanager\plugin__rqt_graph__RosGraph__1\dock_widget__RosGraphUi\parent=__NoneType__
Default\pluginmanager\plugin__rqt_graph__RosGraph__1\dock_widget__RosGraphUi\dockable=true
```

awesomebytes, 2015-06-04:
Same happened to me and some other users. I needed to kill everything that was running and launch everything again to make rqt_graph work.

dirk-thomas, 2015-06-04:
A SSCCE (Short, Self Contained, Correct (Compilable), Example) would really help to find the actual problem. Please consider providing step-by-step instructions how to reproduce this.

ablasdel, 2015-06-04:
Agreed but I haven't been able to get it to happen consistently.

awesomebytes, 2015-06-04:
Ok, I just got it to reproduce in one easy way.

Shell 1:

    roscore

Shell 2:

    rosrun turtlesim turtlesim_node

Now do Control+Z (send to background) to this shell. Then we can execute:

    rosrun rqt_graph rqt_graph

And it will crash with that problem. I suspect it has something to do with the node not being able to answer (`rosnode ping /turtlesim` fails).

dirk-thomas, 2015-06-08:
@awesomebytes  I can reproduce your use case. But I think your expectations are wrong. Ctrl-Z does *not* send the process to background - it suspends the process. And therefore it can not respond to any requests anymore.

If you run `bg %jobnum` after pressing Ctrl+Z the process continue to run and you should be able to ping it and use it with rqt_graph.

@ablasdel I assume you are having a similar situation where one of the nodes is not responding. But in your case it might be a node doing some long-running operation and is therefore not able to respond.

ablasdel, 2015-06-08:
@dirk-thomas That makes sense I will look into trying to reproduce that knowing that.

Either way rqt dotgraph should probably handle it more gracefully than a trace if it can't communicate with a node right?

dirk-thomas, 2015-06-08:
The better handling has been added in ros-visualization/qt_gui_core@97985bd011287d4017ed07be9dcf4806e01e1298

But it would be good to fix `rqt_graph` to not generate an empty subgraph in the first place.

dirk-thomas, 2015-06-08:
If you uncheck the `hide_debug` option you should see the not-responding node.

dirk-thomas, 2015-06-08:
I don't think `rqt_graph` can do anything about it. The stacktrace will be fixed in the upcoming release 0.2.28 of `qt_dotgraph`.

closed 2015-06-08 by dirk-thomas

ablasdel, 2015-06-09:
awesome thanks dirk!