ros-visualization/rqt_common_plugins #322: rqt_reconfigure not working after update

Tabjones, 2015-06-03:
Hallo, i recently updated my system with apt-get, i notice it updated some ros-indigo packages but didn't
take care which ones (i can provide a list later if you think it is useful).

The problem is that rqt_reconfigure is not working anymore, whenever you run `rosrun rqt_reconfigure rqt_reconfigure` or load the same plugin from within rqt, the widgets loads but it is blank. 
It seems it can't comunicate with the dynamic reconfigurable node anymore. I'm pretty sure the problem lays within rqt and not dynamic_reconfigure because the dynparam command line works as expected and changes the parameters of nodes.

Also i'm pretty sure it is not my node configuration, because i didnt change that and it was working before the update. Plus also openni2_camera dynamic reconfiguration stopped working and it behaves the same as my node: rqt_reconfigure not working, dynparam working.

Any clue on quick fix ?
Thank you

ablasdel, 2015-06-03:
I don't think rqt_reconfigure has been touched this release and I have colleague that uses it regularly. I'll have her update to the latest packages and see if she gets the issue as well so we can reproduce.

when you get a chance please post the versions you are using as well.

cottsay, 2015-06-03:
@ablasdel is right; nothing has been changed in recent days. I just built the `rqt` packages, `dynamic_reconfigure` and `hokuyo_node` from the latest versions in git, and everything worked as expected.

I'll be able to grab an openni2 device tonight to test with, but I doubt it will perform any different.

Tabjones, 2015-06-04:
`ros-indigo-rqt-reconfigure  version: 0.3.11-0trusty-20150523-1305-+0000`
`ros-indigo-dynamic-reconfigure version: 1.5.38-0trusty-20150523-0846-+0000`

Interesting, so there must be something wrong with my system, other thing i can think about is the nvidia graphics drivers, also updated those.

Could it be that rqt has some problem rendering the window due to driver updates ?
Is there a way to tell rqt/rviz to use software rendering or opengl rendering, so that i can experiment around those settings ?

I'll look into the docs, thanks for now.

Tabjones, 2015-06-05:
Found an error that may cause this issue:
`Traceback (most recent call last):
  
File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_reconfigure/param_widget.py", line 161, in _filter_key_changed self._nodesel_widget.set_filter(self._text_filter)
  
File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 422, in set_filter self._proxy_model.set_filter(filter_)
  
File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_reconfigure/filter_children_model.py", line 187, in set_filter self.setFilterRegExp(self._filter.get_regexp())

RuntimeError: wrapped C/C++ object of type FilterChildrenModel has been deleted`

This is reported from rqt console when i press the reload button on rqt_reconfigure, widget plugin is still blank.
Any idea what might cause this ?

Tabjones, 2015-06-05:
Found out, i'm not sure what happened, but reinstalling python solved it

closed 2015-06-05 by Tabjones