ros-visualization/rqt_common_plugins #324: Image Creation From Command Line

Bootstrap360, 2015-07-02:
I would like to be able to call a command from the terminal that generates an image of the ROS network and saves it to the location where the command was called.

Being able to set the file name would also be great.

"view_frames" in the tf package has this feature but as PDF. I would prefer an image

ablasdel, 2015-07-02:
There are a number of pdf to image converters out there. If you followed the logic of the tf package's view_frames creation and mimiced it for rqt_graph you could get what you want. (https://github.com/ros/geometry/blob/indigo-devel/tf/scripts/view_frames)

I would be open to a button in rqt_graph that saves the current graph as an image or pdf.

dirk-thomas, 2015-07-02:
`rqt_graph` already has buttons to export to `dot`, `svg` and `png`. I think the feature request is to expose those via the command line.

Since this is a special use case it might be best if you could provide a PR which adds command line arguments to trigger the export.

ablasdel, 2015-07-02:
Oh I didn't know that functionality existed! Thanks dirk!

Bootstrap360, 2015-07-03:
@dirk-thomas 

That is exactly the feature I am looking for.

dirk-thomas, 2017-04-24:
I will close this since it seems the functionality in the GUI was already sufficient.

closed 2017-04-24 by dirk-thomas