ros-visualization/rqt_common_plugins #325: export images from a bag file and save using rqt_bag

indianblood, 2015-07-02:
this option is not available right now

ablasdel, 2015-07-02:
I would be open to a pull requests to add a button to the image_view plugin that saves the currently displayed image: https://github.com/ros-visualization/rqt_common_plugins/blob/groovy-devel/rqt_bag_plugins/src/rqt_bag_plugins/image_view.py

is that what you are thinking?

indianblood, 2015-07-03:
well, I am a beginner to ROS, and, I don't understand what I am supposed to
do with the github code. I have a bag file from which I need to extract
stereo images

ablasdel, 2015-07-03:
If you have a specific problem you need solved I think a better solution is to ask your question over at answers.ros.org. Writing a simple node to dump the topic to a file would probably get you there quicker anyway. 

I still think this enhancement is a good idea and if anyone out there has time to work on it I would be willing to review the pull request.

dirk-thomas, 2015-07-08:
I have created a PR to extend `rqt_image_view` to export images to files: ros-visualization/rqt_common_plugins#326

While that does not solve this specific use case I thought it was a useful first step.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_bag#8

closed 2017-04-24 by dirk-thomas