ros-visualization/rqt_common_plugins #329: Release rqt_common_plugins 0.3.12

ablasdel, 2015-07-23:
- [ ] Create a new indigo-devel branch
- [ ] Release

ablasdel, 2015-07-23:
@dirk-thomas This one actually has some fixes for hydro. Do you think I should release or leave off?
I'm going to look over each commit but I think you probably should too if we are doing hydro here.

dirk-thomas, 2015-07-23:
Your call :wink: If the changes don't pose a high risk for regressions (e.g. a syntax error) you can do a hydro release if it makes sense to make these changes available.

ablasdel, 2015-07-23:
Roger. I have tested all but the rqt_reconfigure personally. @cottsay can you vouch for the rqt_reconfigure changes since the last release? No regressions have become evident right?

closed 2015-07-24 by ablasdel