ros-visualization/rqt_common_plugins #330: Do not hide library symbols

v-lopez, 2015-07-28:
Hiding library symbols takes out the possibility of creating a subclass of a plugin such as rqt_image_view.

In my case I want to create a plugin that allows clicking on an image and publishes the pixel clicked, I either have rebuild all rqt_common_plugins on my repo fork without hiding the symbols, copy the rqt_image_view code  to create the functionality from scratch.

I believe not hiding the symbols would increase the amount of plugins available publicly.

ablasdel, 2015-07-28:
Would you be interested in submitting a PR that exposes the symbols?

DorianScholz, 2015-07-28:
If I understand your question correctly, you want to remove the class_loader_hide_library_symbols call in cmake to be able to link your plugin against the image viewer plugin?

This would not be a good idea for two reasons:

1) Without class_loader_hide_library_symbols all symbols of all libraries linked by image viewer (e.g. libstdc++) would be transitively available to any plugin loaded after the image viewer. This would make unloading of the image viewer impossible. This is why this function exists in the first place...

2) Linking against internal/non-API symbols of another plugin will not be stable against changes to the original plugin, as no one knows, you are using the image viewer plugin as a library.

A better solution would be to move the code you want to reuse into another package (called e.g. rqt_cpp_common) and have image viewer and your code link against it there.
So it can have a defined API and people changing this later on know, that it might be used elsewhere.

This has been done in a similar fashion for the Python plugins in the package rqt_py_common.

v-lopez, 2015-07-28:
Thanks @DorianScholz for the explanation, it makes a lot of sense.

If I were to create rqt_cpp_common and move the reusable code there, after all the reviewing necessary would it be accepted? In other words, it is something that makes sense and will be merged after reviewing.

DorianScholz, 2015-07-28:
Also publishing the mouse click coordinates sounds like a very useful extension.
If all you want to do is extend the image viewer with functionality that might be interesting to others as well, it might be easier to just extend the image viewer and submit your changes as a pull request.

I even think there was talk about this very feature a long time ago.
Hmm, there is a pull request that got some discussion and looks like it could be ready to merge:
https://github.com/ros-visualization/rqt_common_plugins/pull/51
Not sure why it was closed, because as far as I can tell it was not merged...
Maybe @dirk-thomas can shed some light onto this!

DorianScholz, 2015-07-28:
@v-lopez: sure, we are happy to merge pull request!
Not sure what went wrong in the above mentioned case, but usually, its not problem.
Just make sure to follow the code style: http://wiki.ros.org/rqt/Tutorials/Create%20your%20new%20rqt%20plugin#Coding_rule_for_rqt

Maybe you can simply adapt #51 to the current code as a first step, because it was basically complete, but now due to some other changes in the code can not directly be merged at the moment...

v-lopez, 2015-07-28:
Sure I'll look into that, as well as wait for what @dirk-thomas has to say about it.

dirk-thomas, 2015-08-26:
The new PR #331 for publishing mouse events looks good to me. But I haven't had the time to test it. If someone has the time to test it it can be merged.

Since this ticket is about making all symbols of plugins visible I will mark it as closed since we can't do that for the reasons  @DorianScholz described above.

closed 2015-08-26 by dirk-thomas