ros-visualization/rqt_common_plugins #332: rqt_bag: Requires core to be open just to examine bag file

DLu, 2015-08-12:
I believe with rxbag you could just open bag files to examine them without a core. However, rqt_bag requires a core, even without trying to publish anything.

dirk-thomas, 2015-08-12:
In rqt the type of a plugin determines if it requires a ROS node or not. The infrastructure currently does not allow to start as non-ROS-node-plugin and then request a ROS node afterwards. Therefore it is currently not possible to implement the requested feature in a single plugin.

I would be possible though to refactor rqt_bag into two plugins: one which does not allow to publish and one with the current functionality. Not a nice approach though.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_bag#9

closed 2017-04-24 by dirk-thomas