ros-visualization/rqt_common_plugins #345: rqt_bag clock issue

bricerebsamen, 2015-11-14:
I'm trying to use `rqt_bag` to play back data from my bags. I start it with the `--clock` option. It seems to be publishing the clock. However in rviz, the `ROS Elapsed` value keeps resetting to 0: it goes up a bit then resets to 0 and so on. The display looks jerky. I checked #102 and it seems I am doing what I should be doing. I confirm I have `use_sim_time` set to `true`.

Also the terminal gives me those messages:

```
[WARN] [WallTime: 1447468167.169199] [1446107784.144437] Could not process inbound connection: Internal error processing topic [/clock]{'message_definition': '# roslib/Clock is used for publishing simulated time in ROS. \n# This message simply communicates the current time.\n# For more information, see http://www.ros.org/wiki/Clock\ntime clock\n\n', 'callerid': '/rqt_gui_py_node_7572', 'tcp_nodelay': '0', 'md5sum': 'a9c97c1d230cfc112e270351a944ee47', 'topic': '/clock', 'type': 'rosgraph_msgs/Clock'}
[WARN] [WallTime: 1447468167.171413] [1446107784.144437] Could not process inbound connection: [/rqt_gui_py_node_7572] is not a publisher of [/clock]. Topics are [['/rosout', 'rosgraph_msgs/Log']]{'message_definition': '# roslib/Clock is used for publishing simulated time in ROS. \n# This message simply communicates the current time.\n# For more information, see http://www.ros.org/wiki/Clock\ntime clock\n\n', 'callerid': '/rostopic_7732_1447467755958', 'tcp_nodelay': '0', 'md5sum': 'a9c97c1d230cfc112e270351a944ee47', 'topic': '/clock', 'type': 'rosgraph_msgs/Clock'}
[WARN] [WallTime: 1447468168.186813] [1446107784.144437] Could not process inbound connection: [/rqt_gui_py_node_7572] is not a publisher of [/clock]. Topics are [['/rosout', 'rosgraph_msgs/Log']]{'message_definition': '# roslib/Clock is used for publishing simulated time in ROS. \n# This message simply communicates the current time.\n# For more information, see http://www.ros.org/wiki/Clock\ntime clock\n\n', 'callerid': '/rostopic_7732_1447467755958', 'tcp_nodelay': '0', 'md5sum': 'a9c97c1d230cfc112e270351a944ee47', 'topic': '/clock', 'type': 'rosgraph_msgs/Clock'}
```

trainman419, 2015-11-14:
This was pretty broken the last time I tried it; particularly for bag files with lots of topics. See #244

v01d, 2015-11-16:
I'm playing a bag with four topics (it is a 30gb bag, it contains images).
The clock is jumping back every now and then, and this obviously causes issues:

clock: 
  secs: 1225720049
  nsecs: 904932000
---
clock: 
  secs: 1225720050
  nsecs: 37400007
---
clock: 
  secs: 1225720050
  nsecs: 54924000
---
clock: 
  secs: 1225720050
  nsecs: 37400007
---
clock: 
  secs: 1225720050
  nsecs: 54924000
---
clock: 
  secs: 1225720050
  nsecs: 204922000

ablasdel, 2015-11-16:
This clock issue is on the top of the list when I get some rqt developer clock cycles (either my own or someone else willing to help out)

ablasdel, 2015-11-16:
Has anyone tried out #340?

I have yet to test it or but he claims to have solved part of this issue.
Not sure this is the right approach but it might help you guys at least.

ablasdel, 2015-11-17:
closed by #340

closed 2015-11-17 by ablasdel

HWiese1980, 2018-01-22:
Wondering why this issue has been closed. I'm still having some major issue with the simulated clock. After only a couple of messages the publishing on /clock stops. I can see this behavior in `rviz` where the `ROS Time:` clock just stops and also by means `rostopic echo /clock`. `use_sim_time` is true, `rqt_bag` is started with `--clock` parameter. 

Seems to especially happen when the playback of the bag file resets, i.e. restarts from the beginning. But I've also seen this happen somewhere in the midst of the playback.

dirk-thomas, 2018-01-22:
This issue has been closed because the referenced PR reportedly fixed the described problem. If you are experiencing a similar / different / same issue please consider creating a new issue for it. Note that the new location of this plugin is now [ros-visualization/rqt_bag](https://github.com/ros-visualization/rqt_bag). If the same issue happens with the command line version of `rosbag` it might be more appropriate to fill an issue in [ros/ros_comm](https://github.com/ros/ros_comm) instead.