ros-visualization/rqt_common_plugins #351: rqt_topic: float division by zero

bchretien, 2016-02-10:
I encountered the following error while replaying a ros bag that publishes the clock time:

```txt
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_topic/topic_widget.py", line 185, in refresh_topics
    self._update_topics_data()
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_topic/topic_widget.py", line 196, in _update_topics_data
    bytes_per_s, _, _, _ = topic_info.get_bw()
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_topic/topic_info.py", line 111, in get_bw
    bytes_per_s = total / (rospy.get_time() - self.timestamps[0])
ZeroDivisionError: float division by zero
```

The error happens [there](https://github.com/ros-visualization/rqt_common_plugins/blob/master/rqt_topic/src/rqt_topic/topic_info.py#L111). We should check that `dt > 0`.

ablasdel, 2016-02-10:
Seems like the get_bw function could return None * 4 in the case bytes_per_second would essentially be infinite.

ablasdel, 2016-02-10:
Just realized that on some systems get_time will have second level resolution so this should definitely be fixed.

PR #352 (2016-02-23): check for divide by zero and data failures
Fixes #351
closed 2016-02-23 by ablasdel