ros-visualization/rqt_common_plugins #353: ros_graph (and other plugins) end with seg fault

B3J4y, 2016-02-18:
When I try to use for example ros_graph after starting it takes some time and then just return a seg fault. I tried many different possibilities to start 1. "rosrun rqt_graph rqt_graph", 2. just "rqt_graph" on command line and 3. open rqt and then try to choose the graph plugin.

It doesn't give me any other output. So if I can give you the log, plz give me a note about the destination.

The same behaviour occurs at other plugins.

dirk-thomas, 2016-02-19:
The `rqt_graph` plugin is Python only. So a segfault is very weird. Can you please post the full output you are seeing when the UI crashes.

B3J4y, 2016-02-20:
One message was "Cannot access memory at address 0x254716 PC(0x254716) not located in a known VMA region (needed executable region)!" at the Visualization>>Plot Plugin. After that test it just gave me the segfault back.

Can you conclude something from this message? If you tell me where I can find the log then I can give your its output.

dirk-thomas, 2016-04-12:
I will close this ticket for now due to lack of information. Please feel free to comment here with more information and this can be reopened.

closed 2016-04-12 by dirk-thomas