ros-visualization/rqt_common_plugins #356: rqt_dep: display package info from different distro?

scpeters, 2016-03-08:
We are approaching releasing season for ROS K, and a common question is which packages are blocking the release of other packages. The `rqt_dep` tool is great for visualizing package dependencies. It would be great if we could use this tool to visualize which packages have been released into a new ROS distro, to identify the blockages.

In discussing this in the past, a known complicating factor is that dependencies may change from one ROS distro to the next, so this can be an ill-posed problem in some cases.

Most of the time, though, the dependencies don't change from one distro to the next. As a first step, I would load `rqt_dep` with the most recently released ROS distro (jade at the moment) and would like to select another distro from a drop-down combo-box and then colorize the graph.

Any hints are welcome.

dirk-thomas, 2016-03-08:
That sound's like an interesting extension. Currently `rqt_dep` only queries the state of the packages on the system using `rospkg`. The feature you describe would need to use `rosdistro` to have access to packages not on the system as well as other distributions.

It would be great if you could work on extending this plugin or creating a new plugin which uses `rosdistro` for this purpose.

wjwwood, 2016-03-08:
I think that `rqt_dep` only operates on what packages you have locally. So, you need to have all of the packages in Indigo installed to get `rqt_dep` to see them. So the first thing you'd want to do is make `rqt_dep` or a new tool like `rqt_dep` able to pull information from the `rosdistro` cache. This is a file that our build farm generates very frequently which contains all the released packages for a distribution. It also includes the `package.xml` files for each of those released packages. This is how the script that generates the wiki page works:

https://github.com/ros-infrastructure/ros.org_scripts/blob/master/wiki.ros.org/jade_metapackage_status.py

So it would be a marriage of the above script's data access and `rqt_dep`'s graph visualization logic.

scpeters, 2016-03-08:
Thanks, that should be enough to get started.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_dep#2

closed 2017-04-24 by dirk-thomas