ros-visualization/rqt_common_plugins #361: rqt_bag does not play all messages in dense topics

curranw, 2016-04-08:
I'd like to visualize a robot performing a specific task. I load the bag in rqt_bag, publish tf, and set "Play all Messages".

When visualizing the robot description in RViz the robot is "teleporting" around the space. I looked into this, and found that not all of the tf messages are being published, in spite of setting "Play all Messages". When using rosbag play, this problem doesn't occur.

To test this, I subscribed to tf when playing the bag in rqt_bag and rosbag play. I output the messages to two separate files and compare. Many messages played in rosbag play don't show up in rqt_bag. When I do a wordcount on "header", there were around 3,000 tf messages for rqt_bag and 20,000 for rosbag. A huge difference.

Is there a current workaround for this?

trainman419, 2016-04-08:
rqt_bag is based on the python API to rosbag, and in some situations it's simply too slow to play back dense bag files well.

I've played with similar situations in the past, and always ended up switching back to the command line rosbag or the C++ API for playing dense bag files.

curranw, 2016-04-11:
That's a bit disappointing. I added some bag annotation functionality to rqt_bag, but it's difficult to annotate a robot's movements when it's teleporting around. I'll snoop around a bit more and see if there's a way to work around this.

Could this be an issue specific to tf topics? If I use rosbag info it says tf contains 2700 messages at 100Hz, which is accurate. If I use rqt_bag it plays approximately that many messages. However, if I do a rosbag play it actually publishes 20,000 messages. I'm **guessing** this is because there can be many tf messages at the same time stamp. Maybe this is an issue with rosbag rather than rqt_bag?

trainman419, 2016-04-11:
I don't think you're counting messages correctly; each tf2_msgs/TFMessage is an array of geometry_msgs/TransformStamped, each of which may contain a header.

Some nodes (such as robot_state_publisher) can publish many transforms in a single TFMessage, which could cause your counts to be off by quite a bit.

curranw, 2016-04-11:
That's true. If I grep for "transforms" rather than "header" I there are a similar number of messages. 

This fixed my issue, although it's kind of a hack: http://answers.ros.org/question/11477/using-tf-data-from-bag-files/

closed 2016-04-11 by curranw