ros-visualization/rqt_common_plugins #363: rqt_bag error on kinetic

jgoppert, 2016-04-20:
```bash
$ rqt_bag
Traceback (most recent call last):
  File "/home/jgoppert/git/catkin/devel/bin/rqt_bag", line 6, in <module>
    exec(fh.read())
  File "<string>", line 5, in <module>
  File "/home/jgoppert/git/catkin/devel/lib/python2.7/dist-packages/rqt_bag/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 35, in <module>
  File "/home/jgoppert/git/catkin/src/rqt_common_plugins/rqt_bag/src/rqt_bag/plugins/topic_message_view.py", line 34, in <module>
    from python_qt_binding.QtGui import QAction, QIcon, QToolBar
ImportError: cannot import name QAction
```

dirk-thomas, 2016-04-20:
The package has not yet been released into Kinetic yet (since it requires to be updated to work with Qt 5). You can follow the ongoing effort on the `qt5` branch: https://github.com/ros-visualization/rqt_common_plugins/tree/qt5

closed 2016-04-20 by dirk-thomas

jgoppert, 2016-04-20:
Great, thanks.

jgoppert, 2016-04-20:
I'll use that branch and try to give you some good feedback on qt5.

dirk-thomas, 2016-04-20:
You will likely also need the latest changes on these repos: ros-visualization/python_qt_binding#37 ros-visualization/qt_gui_core#65

jgoppert, 2016-04-20:
@dirk-thomas With those branches rqt_bag works for me and I'm able to replay and view images from a log file.

dirk-thomas, 2016-04-20:
Great to know. Are you using `PyQt5` on Xenial?