ros-visualization/rqt_common_plugins #373: rqt_graph fails to launch on jessie due to missing dependency on python-pyqt3.qtsvg

tfoote, 2016-05-06:
I suspect that this should be fixed in one of the dependencies such as rqt_gui, python_qt_bindings or qt_gui but I'm not sure where is the correct place. Or maybe a rosdep rule just needs to be extended. 

Traceback: 
```
linaro@linaro-alip:~$ rqt_graph 
RosPluginProvider.load(rqt_graph/RosGraph) exception raised in __builtin__.__import__(rqt_graph.ros_graph, [RosGraph]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 77, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 39, in <module>
    from python_qt_binding.QtSvg import QSvgGenerator
ImportError: No module named QtSvg

PluginManager._load_plugin() could not load plugin "rqt_graph/RosGraph":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 83, in load
    raise e
ImportError: No module named QtSvg

```

Installing `python-pyqt5.qtsvg` lets it run w/o error.

dirk-thomas, 2016-05-06:
The runtime dependency on Qt is coming from there: https://github.com/ros-visualization/python_qt_binding/blob/621fc06a28370189537b8cc88eb03c7434be61e3/package.xml#L31

So you might want to adjust the rosdep key `python-qt5-bindings` for Jessie. Let me know when the rosdep rule has been updated then I can do a debinc release.

tfoote, 2016-05-06:
PR in #11435

dirk-thomas, 2016-05-06:
Do you expect more fixes for Jessie or should I do a debinc release now?

tfoote, 2016-05-06:
There might be some more subpackages necessary. I see a bunch of pyside extensions but not pyqt when I install on xenial. I'll try to test more of the plugins sometime next week.

dirk-thomas, 2016-05-09:
The new rosdep rule is being used by 0.3.1-2 of python_qt_binding: ros/rosdistro#11460

closed 2016-05-09 by dirk-thomas

acbuynak, 2020-01-27:
Hello! I've recreated this same error statement with the missing QtSvg module in ROS Melodic (ubuntu bionic) and have not been able to resolve my issue by reinstalling python-pyqt5.qtsvg, rqt_common_plugins, or rqt_graph. Do you have any further suggestions where I might look for the issue source?
Thanks!

dirk-thomas, 2020-01-27:
> Hello! I've recreated this same error statement with the missing QtSvg module in ROS Melodic (ubuntu bionic) and have not been able to resolve my issue by reinstalling python-pyqt5.qtsvg, rqt_common_plugins, or rqt_graph. Do you have any further suggestions where I might look for the issue source?
Thanks!

Cross post: https://answers.ros.org/question/342476/missing-qtsvg-module-depended-upon-by-rqt_graph/