ros-visualization/rqt_common_plugins #375: rqt_graph text labels for nodes and topics seem too large in kinetic/Xenial

kengregson, 2016-05-12:
png image attached

[image]

dirk-thomas, 2016-05-12:
This is likely unrelated to Kinetic / Xenial - I would expect you to see the same on Indigo / Trusty - especially since the plugin uses the same code for both. I can reproduce the problem on a system which has a high resolution monitor which uses a factor of around 2 for "Scale for menus and title bars". Can you confirm that you have configure a system wide scaling factor on your system?

dirk-thomas, 2016-05-13:
The only workaround I am currently aware of is setting `QT_DEVICE_PIXEL_RATIO=2` in your environment before starting `rqt_graph`. Can you please try that and comment with the result.

kengregson, 2016-05-13:
Yes, 14" QXD+ display on a Razer Blade with system scale factor set to
1.5x.  Good catch, I'll check with 1x to confirm.

This is likely unrelated to Kinetic / Xenial - I would expect you to see
the same on Indigo / Trusty - especially since the plugin uses the same
code for both. I can reproduce the problem on a system which has a high
resolution monitor which uses a factor of around 2 for "Scale for menus and
title bars". Can you confirm that you have configure a system wide scaling
factor on your system?

kengregson, 2016-05-14:
Confirmed QT_DEVICE_PIXEL_RATIO=2 scales rqt_graph text as desired (with system wide scaling factor set to 1.5 for high resolution monitor).  Before and after attached.  Thanks!
[image]
[image]

closed 2016-05-14 by kengregson