ros-visualization/rqt_common_plugins #378: rqt_bag: Scrolling on timeline view causes crash

mjcarroll, 2016-05-17:
Using 16.04 x64, QT 5.5.1, Kinetic:

Loading a bag in rqt_bag then using the mouse scroll wheel over the timeline causes the crash:

```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 496, in on_mousewheel
    self._timeline_frame.on_mousewheel(event)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 1090, in on_mousewheel
    dz = event.delta() / 120.0
AttributeError: 'QWheelEvent' object has no attribute 'delta'
```

Looking at this: http://doc.qt.io/qt-5/qwheelevent.html#angleDelta, it looks like `delta` has been deprecated in favor of `angleDelta` and `pixelDelta`.

dirk-thomas, 2016-05-17:
Duplicate of #376. The is already available in the latest release of this package: version 0.4.1.

closed 2016-05-17 by dirk-thomas

mjcarroll, 2016-05-18:
Thanks and apologies, I even searched (poorly)!