ros-visualization/rqt_common_plugins #380: rqt_bag:thumbnail toggle causes crash

hugbed, 2016-06-13:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 280, in paint
    self._draw_topic_histories(painter)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 384, in _draw_topic_histories
    self._draw_topic_history(painter, topic)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_frame.py", line 464, in _draw_topic_history
    renderer.draw_timeline_segment(painter, topic, stamp_start, stamp_end, region_x_start, msg_y, region_width, msg_height)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag_plugins/image_timeline_renderer.py", line 130, in draw_timeline_segment
    painter.setPen(QPen(QBrush(Qt.black)))
TypeError: QPen(): argument 1 has unexpected type 'QBrush'
```

Toggling the thumbnails in rqt_bag causes a crash in image_timeline_renderer.py.
Seems like the constructor of `Qt.black` should have Qt.black and not QBrush(Qt.black).

dirk-thomas, 2016-06-27:
Addressed by #381.

closed 2016-06-27 by dirk-thomas

ubuntuslave, 2016-08-11:
I'm getting the same issue on Ubuntu 16.4 and ROS kinectic. Is the solution install from source?