ros-visualization/rqt_common_plugins #383: rqt_graph crashed:  triangulation failed

willdzeng, 2016-06-25:
PluginHandlerDirect._restore_settings() plugin "rqt_graph/RosGraph#0" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/jade/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 116, in _restore_settings
    self._plugin.restore_settings(plugin_settings_plugin, instance_settings_plugin)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 202, in restore_settings
    self._refresh_rosgraph()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 226, in _refresh_rosgraph
    self._update_graph_view(self._generate_dotcode())
  File "/opt/ros/jade/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 253, in _generate_dotcode
    quiet=quiet)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rqt_graph/dotcode.py", line 568, in generate_dotcode
    dotcode = dotcode_factory.create_dot(dotgraph)
  File "/opt/ros/jade/lib/python2.7/dist-packages/qt_dotgraph/pydotfactory.py", line 160, in create_dot
    dot = graph.create_dot()
  File "/usr/lib/python2.7/dist-packages/pydot.py", line 1802, in <lambda>
    lambda f=frmt, prog=self.prog : self.create(format=f, prog=prog))
  File "/usr/lib/python2.7/dist-packages/pydot.py", line 2023, in create
    status, stderr_output) )
InvocationException: Program terminated with status: -11. stderr follows: Error: trouble in init_rank
	%0 2
	%0 6
	%0 1
	%0 8
	%0 6
	%0 1
	%0 1
	_rtabmap_rtabmapviz 1
	%0 1
	%0 1
	_rtabmap_rgbd_odometry 1
	%0 1
	%0 2
	%0 1
	%0 1
	_rtabmap_rtabmap 1
	_camera_points_xyz_planner 1
	%0 1
	%0 1
	%0 1
	%0 1
	%0 1
	%0 3
	_camera_obstacles_detection 1
	%0 1
	%0 3
	%0 1
	%0 1
	_camera_obstacle_detection_nodelet 4
	%0 2
	_camera_zed_wrapper_node 3
	%0 1
	%0 1
	%0 2
	%0 1
	%0 1
	%0 1
	%0 2
	%0 1
	%0 1
	%0 1
libpath/shortest.c:324: triangulation failed
libpath/shortest.c:192: source point not in any triangle
Error: in routesplines, Pshortestpath failed

dirk-thomas, 2016-06-25:
Please provide information how to reproduce the problem.

willdzeng, 2016-06-25:
Hi,
I am trying to simplify the node to find out which node combination is causing this problem. 
Because I have many node running at the same time, the rqt_graph is really complicate, I will attach screen shots later

dirk-thomas, 2016-09-08:
I will close this ticket for now due to now response. Please feel free to comment with more information and it can be reopened.

closed 2016-09-08 by dirk-thomas

willdzeng, 2016-09-08:
Sure, I tried to reproduce it, but it didn't happen anymore. I will reopen this if it happened again.
Thanks