ros-visualization/rqt_common_plugins #384: rqt_topic catkin_make error

imprs, 2016-07-01:
Hello,
       I am not sure what's missing here. You can take a look here:

==> Processing catkin package: 'rqt_topic'
==> Building with env: '/opt/ros/indigo/env.sh'
==> cmake /home/pi/ros_catkin_ws/src/rqt_common_plugins/rqt_topic -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/rqt_topic -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/home/pi/ros_catkin_ws/build_isolated/rqt_topic'
-- Using CATKIN_DEVEL_PREFIX: /home/pi/ros_catkin_ws/devel_isolated/rqt_topic
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/pi/ros_catkin_ws/build_isolated/rqt_topic/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
CMake Error at /opt/ros/indigo/share/catkin/cmake/assert.cmake:17 (message):

  Assertion failed: check for file existence, but filename
  (RT_LIBRARY-NOTFOUND) unset.  Message: RT Library

Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/tools/rt.cmake:42 (assert_file_exists)
  /opt/ros/indigo/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/ros_catkin_ws/build_isolated/rqt_topic/CMakeFiles/CMakeOutput.log".
See also "/home/pi/ros_catkin_ws/build_isolated/rqt_topic/CMakeFiles/CMakeError.log".
<== Failed to process package 'rqt_topic':
  Command '['/opt/ros/indigo/env.sh', 'cmake', '/home/pi/ros_catkin_ws/src/rqt_common_plugins/rqt_topic', '-DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/rqt_topic', '-DCMAKE_INSTALL_PREFIX=/opt/ros/indigo', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/pi/ros_catkin_ws/build_isolated/rqt_topic && /opt/ros/indigo/env.sh cmake /home/pi/ros_catkin_ws/src/rqt_common_plugins/rqt_topic -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/rqt_topic -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'

Command failed, exiting.

Thanks.

dirk-thomas, 2016-07-02:
You might want to ask your question on answers.ros.org. You should also provide more information about the platform you are using.

imprs, 2016-07-02:
I have already done that, waiting for some solution. I am on Raspbian Jessie :)

dirk-thomas, 2016-07-02:
Can you please post the link to the question. And reference this ticket in the question.

imprs, 2016-07-02:
This is the link to the question on answers.ros.org:

http://answers.ros.org/question/238391/rqt_topic-catkin_make-error-of-rt_library/

closed 2016-09-08 by dirk-thomas