ros-visualization/rqt_common_plugins #387: rqt_console should have auto-resizeable Message column

lewisanderson, 2016-07-14:
Currently, when you launch rqt_console, the Message column is very small (same size as other columns). Messages are nearly always too large for this, which means that each time you launch rqt_console, you need to manually resize the field to be larger. Also, rqt_console usually launches with a window large enough that there is a bunch of wasted space on the right side of the Message table. This is quite frustrating if you are frequently launching rqt_console.

I believe a better default user experience would be to automatically stretch the Message column so that the table fills the space. This way, the Message column is more likely to be wide enough to handle the data present, and if it isnt, the user can very quickly expand the size of the entire window, and the Message column will automatically get larger at the same time.

I propose implementing this by adding this to __init__ in https://github.com/ros-visualization/rqt_common_plugins/blob/master/rqt_console/src/rqt_console/console_widget.py : 
        _self.table_view.horizontalHeader().setResizeMode(1, QHeaderView.Stretch)_

When implemented this way, rqt_console provides an easy, effective user experience by default. You can still manually resize other columns, and Message will shrink to fit, and these sizes will be respected if you expand or shrink the entire window. Also, if you click the "fit columns" button, it will jump to a tight fit on everything, so that still works.

Example of current (no stretch) default behavior:
[image]

Example of default behavior with stretch added:
[image]

Example of behavior with stretch, after manually expanding the entire window:
[image]

Example of behavior with stretch, after manually resizing the Severity, Node, and Stamp columns:
[image]

dirk-thomas, 2016-07-14:
That sounds like a good improvement. Please create a pull request with the proposed change.

lewisanderson, 2016-07-14:
I submitted a pull request, thanks!

https://github.com/ros-visualization/rqt_common_plugins/pull/388

PR #398 (2016-08-08): rqt_console: make message column stretch
Fixes #387.
closed 2016-08-08 by dirk-thomas