ros-visualization/rqt_common_plugins #394: ImportError: No module named numpy

romdos, 2016-08-06:
Hello everyone! I have the next trouble:
when I run rqt_bag bag-file
I get the next error:
RosPluginProvider.load(rqt_bag_plugins/BagPlotPlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.plot_plugin, [PlotPlugin]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 77, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag_plugins/plot_plugin.py", line 35, in <module>
    from .plot_view import PlotView
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag_plugins/plot_view.py", line 77, in <module>
    from rqt_plot.data_plot import DataPlot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/__init__.py", line 34, in <module>
    import numpy
ImportError: No module named numpy

How can I fix it? I tried pip install numpy but no luck. Any help!

romdos, 2016-08-06:
sudo pip install numpy solved this issue

dirk-thomas, 2016-08-08:
It looks like that `numpy` was added in #239 but the dependency was not declared in the manifest. I created #396 to fix that.

In general you should not install dependencies using `pip` but via `apt` instead (in this case `python-numpy`). Simply because your apt packages can be updated as part of a declared dependency and the pip packages won't get updated and even overlay a potentially newer version installed from Debian packages. So using pip package is asking for trouble in the future.

closed 2016-08-08 by dirk-thomas