ros-visualization/rqt_common_plugins #402: rqt_plot topic syntax is ambiguous "Invalid topic spec"

sloretz, 2016-10-20:
rqt_plot uses slashes to specify the value in a published message that should be plotted. This syntax leads to the error "PlotWidget.update_plot(): error in rosplot: Invalid topic spec [...]" in cases like the following

To reproduce

1. save the following as simple_pub.py and make it executable with chmod
 ```
#!/usr/bin/env python

import rospy
import std_msgs
import sys
import random
rospy.init_node("simple_pub", anonymous=True)
pub = rospy.Publisher(sys.argv[1], std_msgs.msg.Float32, queue_size=10)
r = rospy.Rate(10)
while not rospy.is_shutdown():
        r.sleep()
        msg = std_msgs.msg.Float32()
        msg.data = random.random()
        pub.publish(msg)
 ```

2. open 4 terminals, one with each command
 * roscore
 * ./simple_pub.py imu/data
 * ./simple_pub.py imu/data/processed
 * rqt_plot

3. Type the following into the topic box and press the green + to add them to the plot
 1. /imu/data/data
 2. /imu/data/processed/data

4. rqt_plot will correctly plot the first topic, /imu/data/data. The second topic will not be plotted, and the terminal from which rqt_plot was started will spame
 ```
PlotWidget.update_plot(): error in rosplot: Invalid topic spec [/imu/data/processed/data]: 'Float32' object has no attribute 'processed'
 ```

This can be fixed by changing the character used to separate message fields. I think the ':' character is a good choice
* /imu/data:data
* /imu/data/processed:data

dirk-thomas, 2016-10-20:
The choice of using the same character as a namespace as well as field separator is certainly a bad one. But that decision was made at a very early time in ROS and is implemented in some of the core ROS packages. So I don't think this package can do much about it if it doesn't want to diverge from the "standard". I am not sure how feasible it will be to change this in the core.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_plot#7

closed 2017-04-24 by dirk-thomas