ros-visualization/rqt_common_plugins #411: rqt_graph: Auto-refresh graph

danielsnider, 2016-12-11:
I think rqt_graph would be even better if it had the option to automatically refresh the graph view, showing changes as they happen. I want to be able to monitor rqt_graph to see what is running at any given time without continually clicking refresh. 

- Do you like the auto-refresh idea?
- Do you think auto-refresh should be the default behaviour?
- Do you think auto-refresh should result in the refresh button being removed?
- Do you think auto-refresh should be triggered by nodes state changes or a refresh rate in hz?

I might create a pull request if there are no major concerns. 

Thanks!

dirk-thomas, 2016-12-11:
Having the option to automatically refresh sounds good.

Since there are still use cases where that is not desired the manual mode should still be possible. I am not sure which default mode makes more sense but if that setting is restored on next startup it becomes a one time change for the user if he prefers the opposite.

Since there is no notification mechanism in ROS i think you will need to poll the information in regular intervals.

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_graph#5

closed 2017-04-24 by dirk-thomas