ros-visualization/rqt_common_plugins #413: rqt_bag ROSBagFormatException problem

KitKat7, 2016-12-15:
If ```Publish``` or ```View``` is selected in the newest ```rqt_bag``` (0.4.3), an exception will occur when ```Playing``` (under Ubuntu 14.04, ROS indigo 1.11.20, tried in two computers with the same environments).
Following are the typical exception messages (the data in the rosbag consists of imu and PointCloud2):
```
Exception in thread Thread-11:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 83, in run
    msg_data = self._get_message(bag, playhead_position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 96, in _get_message
    msg_data = self.timeline.read_message(bag, position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 480, in read_message
    return bag._read_message(position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 980, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 2453, in seek_and_read_message_data_record
    header = _read_header(f)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 1682, in _read_header
    raise ROSBagFormatException('Error reading header field: expected %d bytes, read %d' % (size, len(header)))
ROSBagFormatException: Error reading header field: expected 38 bytes, read 34

Exception in thread Thread-10:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 83, in run
    msg_data = self._get_message(bag, playhead_position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 96, in _get_message
    msg_data = self.timeline.read_message(bag, position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 480, in read_message
    return bag._read_message(position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 980, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 2453, in seek_and_read_message_data_record
    header = _read_header(f)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 1669, in _read_header
    raise ROSBagFormatException('Error reading header: %s' % str(ex))
ROSBagFormatException: Error reading header: error unpacking uint32: unpack requires a string argument of length 4
```
When the exception appears, none of the the messages will be published.
It shouldn't be the problems in bag files, since the messages can be published by ```rosbag``` in console and ```rqt_bag``` in another computer with an older version of ```rqt_bag```.  ```rosbag reindex``` and ```rosbag fix``` were tried, yet with no help.
**But**, when all the files (```*.py```, ```*.pyc```, etc.) in ```rqt_bag``` are replaced by the ones in an old version (modified on Mar 9, 2016, might be version 0.3.13), the **problem is gone!**
Perhaps, it was caused by some bugs in the newest version (guessing).

dirk-thomas, 2016-12-15:
I would be helpful if you could share the bag file in question in order to reproduce the problem.

KitKat7, 2016-12-15:
Hi @dirk-thomas, thanks for your reply!
Here is the clipped bag file which could cause the same problem: [IMU_PC.bag](https://drive.google.com/file/d/0B5IWzFZLUID_dURHZzh1N1N2S1E/view?usp=sharing).

KitKat7, 2016-12-19:
Also tested in Kinetic 1.12.6, it had the same problem with that bag file. However, replacing parts of the files  in ```rqt_bag``` with old version's could still be useful.

dirk-thomas, 2016-12-20:
I think this problem has been fixed by #412. Can you please try the patch from that PR and confirm that it works for you too.

KitKat7, 2016-12-21:
I've tried the patch and it could solve this problem. Thank you and @jspricke!

closed 2016-12-21 by KitKat7