ros-visualization/rqt_common_plugins #414: rqt_console: TypeError signal has 2 argument(s) but 3 provided

peci1, 2016-12-16:
This week I started getting the following error when starting rqt_console on Indigo:

    TypeError: QAbstractItemModel.dataChanged[QModelIndex, QModelIndex] signal has 2 argument(s) but 3 provided
    Traceback (most recent call last):
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_console/message_proxy_model.py", line 118, in handle_highlight_filters_changed
        self.dataChanged.emit(self.index(0, 0), self.index(self.rowCount() - 1, self.columnCount() - 1), [])

The console doesn't work at all due to this error.

DorianScholz, 2016-12-19:
The signature of dataChanged seems to have changed somewhere between Qt 4.8 and Qt 5.7:
http://doc.qt.io/qt-4.8/qabstractitemmodel.html#dataChanged
http://doc.qt.io/qt-5/qabstractitemmodel.html#dataChanged

The error is showing due to the change made by @dirk-thomas in:
https://github.com/ros-visualization/rqt_common_plugins/commit/5d8265c7ab9bc9d4232f15a6ada6e4623a4930ff

Not sure what the right move is here, maybe Dirk can elaborate on this change.

dirk-thomas, 2016-12-20:
Thanks for reporting the regression. Can you please try and confirm if the fix from #416 makes it work for you again.

DorianScholz, 2016-12-20:
works for me, thx.

dirk-thomas, 2016-12-20:
Thanks for checking!

PR #416 (2016-12-20): fix regression with Qt 4
Fixes #414.
closed 2016-12-20 by dirk-thomas