ros-visualization/rqt_common_plugins #424: "use Python 3 compatible syntax" breaks build

mechatrocity, 2017-01-31:
Trying to run rqt_bag with Python2 gave the following error (sanitized console output for brevity):
> $ rqt_bag
> Loading input.bag
> Traceback (most recent call last):
> ...
>   File ".../lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 152, in add_bag
>     bag_topics = bag_helper.get_topics(bag)
> NameError: global name 'bag_helper' is not defined
> 

`import rqt_bag.bag_helper`
should be
`from rqt_bag import bag_helper`
in order to refer to the function on its own.  This applies throughout.

dirk-thomas, 2017-01-31:
Thank you for reporting this. Regression of #421. Fixed in 2c2487a02c83628a14be7bde318d20249167ffdb.

Can you please confirm that the referenced commit addresses the problem for you.

closed 2017-01-31 by dirk-thomas

130s, 2017-02-01:
(Tried to find an associated pull request but couldn't, so I commented on a commit https://github.com/ros-visualization/rqt_common_plugins/commit/2c2487a02c83628a14be7bde318d20249167ffdb#commitcomment-20700520. The below is just the copy from there).

----

I just came across the error with when I load a .bag file from GUI. Using the source I confirm the issue is fixed.
```
Loading /home/rosnoodle/Desktop/2017-01-30-15-03-19.bag
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_widget.py", line 264, in _handle_load_clicked
    self.load_bag(filename[0])
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_widget.py", line 293, in load_bag
    self._timeline.add_bag(bag)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 152, in add_bag
    bag_topics = bag_helper.get_topics(bag)
NameError: global name 'bag_helper' is not defined
PluginHandlerDirect._shutdown_plugin() plugin "rqt_bag/Bag#1" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 86, in _shutdown_plugin
    self._plugin.shutdown_plugin()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag.py", line 86, in shutdown_plugin
    self._widget.shutdown_all()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_widget.py", line 357, in shutdown_all
    self._timeline.handle_close()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 127, in handle_close
    for topic in self._get_topics():
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 214, in _get_topics
    for topic in bag_helper.get_topics(bag):
NameError: global name 'bag_helper' is not defined

$ rospack find rqt_bag
/opt/ros/indigo/share/rqt_bag
$ dpkg -p ros-indigo-rqt-bag|grep Ver
Version: 0.4.4-0trusty-20170127-144938-0800
```

dirk-thomas, 2017-02-01:
I assume the stacktraces are from before you applied the patch since you mention that the "using the source I confirm the issue is fixed"?

130s, 2017-02-01:
Ah, yes, output in https://github.com/ros-visualization/rqt_common_plugins/issues/424#issuecomment-276741007 is the error I got before without using the patch.

dirk-thomas, 2017-02-01:
Ok, I just wanted to be sure. Thank you for testing this and reporting back.