ros-visualization/rqt_common_plugins #432: rqt_topic crashes

dseredyn, 2017-02-20:
how to reproduce:
1. rosrun rqt_topic rqt_topic
2. select a topic

this gives an error:
[ERROR] [1487596624.452584, 40.374000]: bad callback: <bound method TopicInfo.message_callback of <rqt_topic.topic_info.TopicInfo object at 0x7f94480baed0>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_topic/topic_info.py", line 100, in message_callback
    self.sizes.append(buff.len)
AttributeError: 'cStringIO.StringO' object has no attribute 'len'

dirk-thomas, 2017-02-20:
Thank you for the report. Please see #434 for a fix. It would be great if you could try the proposed patch and confirm that it resolves the problem for you.

khancyr, 2017-02-22:
@dirk-thomas Hi, I confirm that the patch is correcting the problem

dirk-thomas, 2017-02-22:
Thank you for trying and confirming.

PR #434 (2017-02-22): fix buffer length
Fixes #432.
closed 2017-02-22 by dirk-thomas