ros-visualization/rqt_common_plugins #437: rqt_image_view: image shrinks when resolution changes, or has wrong aspect if '1' is enabled

lucasw, 2017-02-28:
When the '1' button is disabled, the first image received will fill the available widget space properly, and a following image with a different resolution may work properly (I think depending on if the vertical resolut, but a third image will be the same width or height as the previous image, and therefore be far smaller than available space, and each following image which changes resolution will also choose the width or height to match which continues to shrink the image.

This continues until some minimal size has been reached, after that the aspect ratio no longer changes, the image is updated in the aspect ratio of that minimal size.

```
#!/usr/bin/env python

import cv2
import numpy as np
import rospy

from sensor_msgs.msg import Image

class AlternateImage:
    def __init__(self):
        self.pub = rospy.Publisher("image", Image, queue_size=4)

        self.height = 1024
        self.width = 768
        self.channels = 3

        image1 = np.zeros((self.height, self.width, self.channels), np.uint8)
        msg1 = Image()
        msg1.encoding = "bgr8"
        msg1.height = self.height
        msg1.width = self.width
        msg1.step = self.width * self.channels
        msg1.data = image1.tostring()

        image2 = np.ones((self.width, self.height, self.channels), np.uint8) * 255
        msg2 = Image()
        msg2.encoding = "bgr8"
        msg2.height = self.width
        msg2.width = self.height
        msg2.step = self.height * self.channels
        msg2.data = image2.tostring()

        while not rospy.is_shutdown():
            msg1.header.stamp = rospy.Time.now()
            self.pub.publish(msg1)
            rospy.sleep(1.0)
            msg2.header.stamp = rospy.Time.now()
            self.pub.publish(msg2)
            rospy.sleep(1.0)

if __name__ == '__main__':
    rospy.init_node("alternate_image")
    alternate_image = AlternateImage()

```

If the '1' button is enabled, then the first image received will define the aspect ratio, then following images will conform to that aspect ratio wrongly, but the images continue to fill the widget.

I tried this with the latest master branch code, looks to behave the same as the Jade release version.

lucasw, 2017-02-28:
I have a 4MB animated gif, but it gives me a `Something went really wrong, and we can’t process that file.` error.

But here is an alternate gif host:

https://plus.google.com/+LucasWalter/posts/WT6mXSy5BUv

dirk-thomas, 2017-04-24:
This issue was moved to ros-visualization/rqt_image_view#3

closed 2017-04-24 by dirk-thomas