ros-visualization/rqt_common_plugins #443: rqt_top/node_info.py

kylerlaird, 2017-03-15:
In /opt/ros/kinetic/lib/python2.7/dist-packages/rqt_top/node_info.py is the line "from xmlrpclib.socket import error".  This causes an exception.

I eliminated the exception by replacing that line with "import xmlrpclib" and "error = xmlrpclib.socket.error".  I am not confident that this is a good solution but it's working for me.

dirk-thomas, 2017-03-17:
Can you please describe a reproducible example which results in the exception. And propose your patch as a pull request (of just point to a patch if you don't think the patch is ready to be merged yet).

SammyRamone, 2017-03-27:
I can confirm this bug. I get the following error when trying to open the rqt_top plugin

> RosPluginProvider.load(rqt_top/TOP)
 exception raised in __builtin__.__import__(rqt_top.top_plugin, [Top]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_top/top_plugin.py", line 36, in <module>
    from rqt_top.node_info import NodeInfo
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_top/node_info.py", line 35, in <module>
    from xmlrpclib.socket import error
ImportError: No module named socket

> PluginManager._load_plugin()
 could not load plugin "rqt_top/TOP":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 86, in load
    raise e
ImportError: No module named socket

I could solve the problem similar to the proposed solution by kylerlaird and made a pull request

It seems that the line was changed to make the plug-in more python3 compatible.

dirk-thomas, 2017-03-27:
Should be fixed by #446.

closed 2017-03-27 by dirk-thomas