ros-visualization/rqt_common_plugins #447: rqt_web does not work with Kinetic/Ubuntu 16.04

mantkiew, 2017-04-11:
Hello, rqt_web no longer works on Kinetic/Ubuntu 16.04. It must be related to moving to Qt5 in kinetic.

```
RosPluginProvider.load(rqt_web/Web) exception raised in __builtin__.__import__(rqt_web.web, [Web]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_web/web.py", line 35, in <module>
    from rqt_web.web_widget import WebWidget
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_web/web_widget.py", line 49, in <module>
    from python_qt_binding.QtWebKit import QWebPage, QWebView
ImportError: No module named QtWebKit

PluginManager._load_plugin() could not load plugin "rqt_web/Web":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 86, in load
    raise e
ImportError: No module named QtWebKit
```

dirk-thomas, 2017-04-11:
It works for me. How did you install the ROS packages - from source or from Debian packages? Have you installed the necessary `webkit` packages. I don't know the exact names but probably `python-pyside2.qtwebkit`, `libqt5webkit5`.

mantkiew, 2017-04-11:
I used `sudo apt-get install ros-kinetic-desktop` + some additional packages. 

```
libqt5webkit5 is already the newest version (5.5.1+dfsg-2ubuntu1).
python-pyside2.qtwebkit is already the newest version (2.0.0+dev-0~201604151742~rev1858~pkg38~ubuntu16.04.1).
```

The rosdeps are installed as well:

```
$ rosdep install rqt_web
#All required rosdeps installed successfully
```

dirk-thomas, 2017-04-11:
Are you using `PySide`? You can check in the `About` dialog of `rqt`.

mantkiew, 2017-04-11:
```
￼
rqt
rqt is a framework for graphical user interfaces. It is extensible with plugins which can be written in either Python or C++.
Please see the Wiki for more information on rqt and available plugins.
Utilized libraries:: Python 2.7.12, rospkg 1.1.0, PyQt 5.5.1 (QtCore, QtGui, QtNetwork, QtOpenGL, QtSvg, QtWidgets, QtXml), Qt 5.5.1, SIP C++ bindings available.
```

dirk-thomas, 2017-04-11:
Since you use PyQt do you have the package `python-pyqt5.qtwebkit` installed?

mantkiew, 2017-04-11:
No, I didn't have that one. After installing it, `rqt_web` now works!
Thanks a lot!

dirk-thomas, 2017-04-11:
Thanks for iterating on this. I updated the title since the dependencies need to be fixed.

dirk-thomas, 2017-04-18:
I added the missing dependency in #449.

PR #449 (2017-04-18): add missing webkit dependency
Fixes #447.
closed 2017-04-18 by dirk-thomas